Ardumower/Sunray #14: error: no map route

jopl, 2021-05-11:
Dear ardumower sunray enthusiastics, I successfully built the ardumower sunray model with standalone RTK. Everything seems to be working during creating the map, but after map is created and calculated, the tracking lines appears in the map and system writes the error "no map route" and rover doesn't want to start mowing. Is it possible to have such issue due to I am using free license, please? And does it solve after purchase of new license? Or what other problem could be the reason for such error message?
Thank you for your answer in advance. Kind regards, jopl.

greymfm, 2021-05-12:
Hello, GitHub is not a support forum, have you considered to ask in the forum? :-) 
https://forum.ardumower.de/forums/sunray-firmware.22006/

jopl, 2021-05-12:
Dear Alex, of course, thank you. It is solved at the moment, the issue was too small test area ... when I use larger one, everything is OK. Thank you so much. Regards, Josef.

closed 2021-05-12 by jopl