Ardumower/Sunray #19: Clockconfig or Cache enabled

Viatorus, 2021-06-01:
I don't see any clock configuration or SAMD51 cache enable instructions inside the sources.

Why is that so? Did I missed it?
At what frequency runs the CPU?

rweickelt, 2021-06-05:
> I don't see any clock configuration or SAMD51 cache enable instructions inside the sources.

The Arduino IDE sets the define `ENABLE_CACHE` and uses the startup code from `${ARDUINO_DATA_PATH}/packages/adafruit/hardware/samd/1.7.2/cores/arduino/startup.c`. Have a look at that file.

> At what frequency runs the CPU?

The Arduino IDE sets the define `F_CPU=120000000L`. Search for `F_CPU` in the very same file.

Viatorus, 2021-06-06:
Ah, thank you! Just wanted to make sure it is not missed here. :)

closed 2021-06-06 by Viatorus