Ardumower/Sunray #22: PWM frequency

felixganzer, 2021-06-27:
Hi,

In der pinman.cpp wurde während des refactorings die TCC_TOP von 0X80 auf 0XFF für den SAMD51 mit brushed Motoren geändert. Damit ändert sich die PWM Frequenz von 3.6 kHz auf 1.8 kHz. War das Absicht oder ein versehen?

greymfm, 2021-06-28:
Hi,
danke vielmals für den Hinweis - anscheinend ist da ein Fehler unterlaufen oder das war Absicht. Hast Du die Frequenz mal gemessen die derzeit heraus kommt? Es sollen 3900 Hz sein (Begründung von damals herausgekramt, die füge ich mal noch zum Code hinzu) .

 Why using 3900 Hz PWM frequency? 
   At 3900 Hz, our MC33926 motor drivers give linear motor current results for all speeds (PWM/odometry values)
	 http://wiki.ardumower.de/images/f/f9/Pwm_3khz.png
	 At 490 Hz or 20 kHz, the reported values are not so good:
	 http://wiki.ardumower.de/images/e/e0/Pwm_490hz.png
	 http://wiki.ardumower.de/images/8/84/Pwm_20khz.png

felixganzer, 2021-06-28:
Hi,

gemessen habe ich das nicht, weil ich nicht das notwendige Equipment zuhause habe. Ich habe nur theoretisch nachgerechnet und die commits im Git nachverfolgt. Laut Forum ist das entstehende Geräusch an den Motoren einigen aufgefallen, mit der Begründung, dass dies bei älteren Softwareversionen nicht da war. Spricht für einen Bug.

greymfm, 2021-06-28:
Danke. Ich habe den TCC_TOP-Wert einfach mal wieder auf den ursprünglichen geändert (https://github.com/Ardumower/Sunray/commits/master) in der Hoffnung dass dies 3900 Hz ergibt :-)

greymfm, 2021-06-28:
Ich habe gerade mal gemessen, funktionierende PWM-Regelung oberhalb von 1800 Hz bekommen ich nur mit: 
    #define TCC_CTRLA_PRESCALER TCC_CTRLA_PRESCALER_DIV64     // PWM base frequency
    #define TCC_TOP 0xFF                                       // PWM resolution
 Dies ergibt ca. 7300 Hz. - dabei belassen wir es erstmal :-)

felixganzer, 2021-06-28:
Ich teste es nachher am lebenden Gerät

felixganzer, 2021-06-28:
Ich habe es kurz getestet. Der Ardumower funktioniert noch und das Pfeifen ist höher frequenter und damit für mich weniger nervig.

closed 2021-06-28 by felixganzer