Ardumower/Sunray #25: robot for a second in-place rotating in the wrong direction, then correcting

greymfm, 2021-08-18:
With activated IMU (not with odometry-only), the robot sporadically in-place rotates for a second in the wrong direction, then corrects and rotates into the right direction.
Idea: log a session with rotation data (IMU angle, set-rotation angle, current-rotation angle etc.) to find out if it's a calculation or maybe hardware issue

stephwe, 2021-08-24:
How can I log a session with rotation data (IMU angle, set rotation angle, current rotation angle)? Is the normal log file, which is saved on the SD card, sufficient or should the logging take place in the serial monitor?
I only noticed the problem now, but yes, it is there.

ShadedSelf, 2025-07-27:
The problem is on this line:
https://github.com/Ardumower/Sunray/blob/30630f14e7f1b5bc109562c92fbf48d9dfe949f9/sunray/LineTracker.cpp#L48

When the mower gets too close to the end point the angle values can swing erratically.
It can be fixed by taking the angle between lastTarget and target.