Ardumower/Sunray #45: experiment with low-pass filtering GPS float solution

greymfm, 2021-09-19:
Currently, robot wobbles around tracked path if using GPS float solution. Try out if low-pass filtering GPS float solution would help.

Idea:
In 'void computeRobotState()', try out a low-pass filter for GPS float solution:
https://github.com/Ardumower/Sunray/blob/d7428c5112738cf8d727167a07c5e956a834f902/sunray/robot.cpp#L931

 if (maps.useGPSfloatForPosEstimation){ // allows planner to use float solution?
   stateX = 0.9 * stateX + 0.1 * posE;
   stateY = 0.9 * stateY + 0.1 * posN;
}

stephwe, 2021-09-22:
Beim kompelieren in IDE bekomme ich mit den Eintragungen folgende Fehlermeldung:
exit status 1
a function-definition is not allowed here before '{' token

ShadedSelf, 2023-08-17:
Is GPS float data normally distributed around the actual antenna?
If so, averaging over time should give a very good aproximation, if it isnt the mower would not wobble but the position estimation would be biased.

I might give this a shot and see what happens.

greymfm, 2023-08-18:
I don't think float is normally distributed around the actual antenna. In fact, the float solution typically happens due to [multipath-errors](http://grauonline.de/wordpress/?page_id=1468), so the signal measurements are longer than in reality. The result is that the path is deviated from the ground truth into some direction. When the RTK is fixed again typically a GPS jump happens...

Example float recordings:
[image]

[image]

ShadedSelf, 2023-08-18:
I see.

Well, I made a couple of changes:

My mower allways goes into invalid for a sec when it loses fix, so it stops, does the whole motor spin up, and then goes, so I added a 10 second timeout for invalid gps, only using imu + odometry, now the mower just keeps going when it goes into invalid and quickly finds either a float or fix solution. Seems accurate enough.

Then the suggested change of mixing odometry and heading with float, I dont use float gps data to calculate heading here, it seems to work pretty well, I dont know if this is just because the imu is better sort term or beacuse mixing those values is actually doing anything though, but seems better than just flloat.

Ill test further, but it seems to have improved things.

disaster123, 2023-08-18:
@ShadedSelf this sounds interesting - i would like to test this as well. Do you already have some commits or a branch ready?

ShadedSelf, 2023-08-18:
Im working with my local dirty files.
Ive made a new [branch](https://github.com/ShadedSelf/Sunray/tree/deadrekon) if you want to test it, hopefully I didnt miss anything.

disaster123, 2023-08-19:
thanks will have a look

ShadedSelf, 2023-09-03:
This is working fine for me, today I had the mower drive on invalid for 5 seconds or so, then switch to float for another 3 seconds and finally go back to fixed and be exactly where it was supposed to be.
I dont really get longer float times, so I cant really test much longer than that, but its definatelly an improvement for me.

Ive merged this into #145 since both share the vector/quaternion library.