Ardumower/Sunray #47: reset GPS receiver after undocking

greymfm, 2021-09-21:
problem:

Ich hatte gestern noch das Phänomen das der Mover mit einem um 30cm Versatz GPS FIX aus der LS kam. Da er damit auch FIX hatte und auch noch innerhalb des Perimeter war störte den Mover das nicht. Er ist dann natürlich überall gegengerammelt.

Ein manueller Reset des GPS Empfängers brachte dann den richtigen FIX.

Kann man vielleicht optinal über die config.h noch einen zeitgesteuerten Reset des GPS Empfängers nach dem undock auslösen?
Also Startbefehl -> 20sek. (Einstellbar über config.h) später Reset des GPS Empfängers. 

Da er ja nach 20sek aus der LS ist, ist dann ja auch immer freie Sicht und guter GPS Empfang. Da kommt dann auch immer der richtige FIX.

solution:
reset GPS receiver after undocking

Svol0, 2021-09-22:
Das mit dem Versatz ist bei mir fast jedes mal so. Ich handhabe es zur Zeit immer so, dass ich ihn mit der Option DOCK_IGNORE_GPS immer bis zum ersten Dockingpunkt fahren lasse, dann stoppe, dann einen GPS-Reset durchführe. Wenn er anschließend einen GPS-FIX hat, kann ich ihn wieder starten und die Position stimmt.

Da ich bei mir neben der großen Hauptmähfläche eine kleinere Mähfläche weit entfernt von der Docking-Station habe, habe ich die Anfahrt zur Docking-Station etwas modifiziert. In der config.h habe ich folgendes ergänzt:
#define DOCK_SLOW_ONLY_LAST_POINTS 2  // slow speed will only be used for the last number of points (0 = all docking points will be reached with slow speed)
Und in der mapp.cpp anstelle von Zeile 1129 (`if (!sim) trackSlow = true;`):
`// only the last dockingpoints (value from "DOCK_SLOW_ONLY_LAST_POINTS") should be done with low speed`
`if ((dockPoints.numPoints > abs(DOCK_SLOW_ONLY_LAST_POINTS)) && 
          (dockPointsIdx < (dockPoints.numPoints - abs(DOCK_SLOW_ONLY_LAST_POINTS))) && (DOCK_SLOW_ONLY_LAST_POINTS != 0) ){`
        `if (!sim) trackSlow = false;`
      `}`
      `else {`
        `if (!sim) trackSlow = true;`
      `}`
Dadurch fährt er alle Dockingpunkte, bis auf die letzten 2 (oder die Anzahl, die man halt einstellt) mit normaler Geschwindigkeit an und reduziert nur für die exakte Positionierung vor der Ladestation. Dieses möchte ich später auch noch für das Undocking verwenden. Also die ersten 2 Positionen aus der Ladestation heraus nur mit langsamer Geschwindigkeit und ODO/IMU bei erreichen des 2ten Dockingpunkts (betrachtet aus Richtung Ladestation) soll er dann warten, GPS-Reseten und anschließend die letzten Dockingpunkte mit normaler Geschwindigkeit, vorwärts und GPS abfahren.

Den GPS-Reset von einem Dockingpunkt abhängig zu machen, hat den Vorteil, dass man diesen Punkt gezielt an einer Position setzen kann, wo man sicher weiß, dass der GPS-Empfang dort gut ist, damit er seinen FIX bekommt.

Gruß Sven

stephwe, 2021-09-22:
Wäre auch eine gute Idee an einem bestimmten, in der config.h einstellbarem, Dockingpunkt den GPS reset ausführen zu lassen. So kann man mit dem Dockingpunkt bestimmen wo der beste GPS Empfang ist bzw. der reset ausgeführt wird und bekommt dann immer ein sauberes Signal.
Ist optimal gerade für Ladestationen an Häuserwänden.
Diese variante ist noch wesentlich sinnvoller anstatt den GPS reset über Zeit auszulösen.

#define reset GPS receiver after undocking true
#define reset_GPS_receiver_LAST_POINTS 2

Svol0, 2021-09-25:
Für den GPS-Reset beim undocking habe ich folgenden Vorschlag (alle Zeilennummernangabe beziehen sich auf den release Stand 1.0.219):
- Erweiterung der config.h mit folgenden Parametern:
```
#define DOCK_SLOW_ONLY_LAST_POINTS  2 // slow speed will only be used for the last number of points (0 = all points will be reached with slow speed)
#define DOCK_POINT_GPS_REBOOT       2 // dockingpoint number (counted from last dockingpoint) where the gps will be rebooted and waited for gps-fix by undocking. 0 = no gps reboot by undocking
```

- Erweiterung der robot.h bei Zeile 147:
```
extern int dockGpsReboot;                     // status for gps-reboot at specified docking point by undocking action
```

- Erweiterung der robot.cpp bei Zeile 1332 (vor `  if (mow)  {  // wait until mowing motor is running`):
```
  // reboot gps by undocking at a specified docking point (please see "DOCK_POINT_GPS_REBOOT" in config.h) //SOew
  if (dockGpsReboot > 0){   // status dockGpsReboot: 0 = off, 1 = reset gps, 2 = wait for gps-fix
    if (dockGpsReboot == 1){
      gps.reboot();   // reboot gps to get new GPS fix
      dockGpsRebootTime = millis() + 10000; // load check timer for gps-fix with 10sec
      dockGpsRebootFixCounter = 0;
      dockGpsReboot = 2;
      CONSOLE.println("robot.cpp  dockGpsReboot - start gps-reboot");
    }
    if (dockGpsRebootTime <= millis()){
      if (gps.solution == SOL_FIXED){
        dockGpsReboot = 0;
        CONSOLE.print("robot.cpp  dockGpsReboot - got gps-fix after ");
        CONSOLE.print(dockGpsRebootFixCounter);
        CONSOLE.println(" sec");     
      }
      else {
        dockGpsRebootTime = millis() + 10000; // load check timer for gps-fix with 10sec
        dockGpsRebootFixCounter += 10;  // add 10 seconds
        CONSOLE.print("robot.cpp  dockGpsReboot - still no gps-fix after ");
        CONSOLE.print(dockGpsRebootFixCounter);
        CONSOLE.println(" sec");     
      }
    }
    // stop mower
    linear = 0;
    angular = 0;        
    mow = false;
  }
```

- Ersetze in der map.cpp die Zeilen 1120 bis 1165 durch folgenden code:
```
// get next docking point  
bool Map::nextDockPoint(bool sim){    
  if (shouldDock){
    // should dock  
    if (dockPointsIdx+1 < dockPoints.numPoints){
      if (!sim) lastTargetPoint.assign(targetPoint);
      if (!sim) dockPointsIdx++;              
      if (!sim) trackReverse = false;
      // only the last dockingpoints (value from "DOCK_SLOW_ONLY_LAST_POINTS") will be done with slow speed //SOew
      if ((dockPoints.numPoints > abs(DOCK_SLOW_ONLY_LAST_POINTS)) && 
          (dockPointsIdx < (dockPoints.numPoints - abs(DOCK_SLOW_ONLY_LAST_POINTS))) && (DOCK_SLOW_ONLY_LAST_POINTS != 0) ){             
        if (!sim) trackSlow = false;
      }
      else {
        if (!sim) trackSlow = true;
      }
      if (!sim) useGPSfixForPosEstimation = true;
      if (!sim) useGPSfixForDeltaEstimation = true;      
      if (!sim) useGPSfloatForPosEstimation = false;    
      if (!sim) useGPSfloatForDeltaEstimation = false;    
      if (!sim) useIMU = true;     // false      
      return true;
    } else {
[... 38 lines cut ...]
        if (!sim) wayMode = WAY_FREE;      
        if (!sim) trackReverse = false;              
        if (!sim) trackSlow = false;
        if (!sim) useGPSfixForPosEstimation = true;        
        if (!sim) useGPSfixForDeltaEstimation = true;
        if (!sim) useGPSfloatForPosEstimation = true;    
        if (!sim) useGPSfloatForDeltaEstimation = true;    
        if (!sim) useIMU = true;    
        return true;
      } else return false;        
    }  
  }
  return false;
}
```

Funktionsbeschreibung:
- Bei Fahrt zur Docking-Station wird über den Parameter Wert von "DOCK_SLOW_ONLY_LAST_POINTS" die Position des Dockingpunktes angegeben (betrachtet aus Richtung Dockingstation), ab welchem mit langsamer Geschwindigkeit (linear = 0,1) weiter gefahren wird. Alle Dockingpunkte vorher werden mit der normalen (setspeed) Geschwindigkeit angefahren. Ein Wert von "Null" bewirkt, dass alle Punkte mit langsamer Geschwindigkeit angefahren werden.
- Bei Fahrt aus der Docking-Station wird über den Parameter Wert von "DOCK_SLOW_ONLY_LAST_POINTS" die Position des Dockingpunktes angegeben (betrachtet aus Richtung Dockingstation), ab welchem das langsame rückwärtsfahren aus der Station beendet wird und der Mäher den Rest der dockingpunkte mit normaler Geschwindigkeit, vorwärts und mit GPS-Unterstützung weiterfährt. Ein Wert von "Null" bewirkt, dass alle Punkte langsam und rückwärts abgefahren werden.
- Zusätzlich kann durch Angabe eines Wertes bei dem Parameter "DOCK_POINT_GPS_REBOOT" bestimmt werden, an welchem Dockingpunkt (betrachtet aus Richtung Dockingstation) ein GPS-Reboot durchgeführt werden soll. Der Mäher wartet dann, bis ein GPS-Fix vorhanden ist, und setzt das undocking dann fort. Ein Wert von "Null" bewirkt keinen GPS-Reboot beim undocking.

HINWEIS:
Dieses konnte leider nicht ausgiebig von mir getestet werden, da mein Mäher gestern einen defekt an einem Antriebsmotor erlitten hat. Ersatz ist bestellt und es wird weiter getestet, sobald das Ersatzteil da ist. Vielleicht kann dieses zwischenzeitlich jemand weiter testen.

Gruß Sven

stephwe, 2021-09-25:
Konnte heute erste Tests machen. GPS Reset klappt super. Alle Daumen hoch!!
Dann hatte ich mal einen langen Dockingweg mit 8 Dockingpunkten angelegt. Da blieb der Mäher bei Punkt 3 (gesehen von der Dockingstation) mit FLOAT stehen. Ich habe ein FLOAT - Timeout von ca. 20sek eingestellt.
Eigentlich müsste er in der 219 doch den gesamten Dockingpfad mit ODO fahren (wenn eingestellt) falls er keinen FIX hat. 
Ich muss das morgen noch mal genau nachstellen, ist nicht so einfach da auch des Öfteren gleich nach der Ausfahrt ein FIX kommt.

Svol0, 2021-09-25:
Danke für den Test und das Feedback.
Wenn Du die kompletten Änderungen an der 219 durchgeführt hast, dann müsstest Du bei "DOCK_SLOW_ONLY_LAST_POINTS" eine 0 (NULL) einstellen, damit er den kompletten Weg nur mit ODO fährt, wenn "DOCK_IGNORE_GPS" auf true gesetzt ist. Allerdings fährt er dann auch alles nur mit langsamer Geschwindigkeit. Sonst versucht er ab dem bei "DOCK_SLOW_ONLY_LAST_POINTS" eingestelltem Punkt mit GPS-Unterstützung weiter zu fahren.
Ich habe dieses mit reingenommen, da ich bei mir eine kleine Teilfläche habe, die weit von der Dockingstation entfernt liegt und der Mäher fast einmal komplett um das Haus und durch einen schmalen Weg fahren muss. Für die Strecke (ca. 60m) mit vielen Wendungen wäre nur ODO zu ungenau.
Ist der Mäher, wenn er wieder ein Fix bekommt automatisch weiter gefahren?

stephwe, 2021-09-25:
Ja, Mäher ist nach FIX weiter gefahren auf dem Dockingpfad.
Und ja, das macht natürlich so Sinn. 
Anderrum, wenn  DOCK_SLOW_ONLY_LAST_POINTS größer 0 dann hebelt es ja DOCK_IGNORE_GPS aus. 
Vielleicht wäre es sinnvoll die Anzahl der Dockingpunkte, die bei DOCK_SLOW_ONLY_LAST_POINTS eingestellt ist bei
DOCK_IGNORE_GPS = true dann mit ODO anzufahren und bei 
DOCK_IGNORE_GPS = false so, wie es jetzt ist

Svol0, 2021-09-25:
Der Parameter "DOCK_IGNORE_GPS" ist normal nur beim Ausfahren aus der Station aktiv. Nicht für das Einfahren in die Station. Dieser Parameter ist von Alexander hinzugefügt worden, da es vorkommen konnte, wenn der Mäher einen falschen GPS-Fix in der Station bekommen hat und Rausfahren soll, der Mäher nicht Rückwärts rausfährt, sondern versucht in der Station zu drehen, was dann nicht gut ausging. Von daher ist es eigentlich nur wichtig, dass er bis zum ersten Dockingpunkt nach der Station ohne GPS fährt. (Vorausgesetzt, der Dockingpunkt liegt weit genug vor der Station, dass der Mäher gefahrlos wenden kann)

Wahrscheinlich ist es besser, den Parameter "DOCK_SLOW_ONLY_LAST_POINTS" nur für die Fahrt zur Docking-Station zu verwenden und der Parameter "DOCK_POINT_GPS_REBOOT" wäre dann automatisch der Wegpunkt, an dem auf das GPS-FIX beim Ausfahren gewartet wird und von wo aus der Mäher seine Reise Vorwärts, mit normaler Geschwindigkeit und GPS-Unterstützung fortsetzt kann.
Bei einem Wert von 1 für "DOCK_POINT_GPS_REBOOT" würde der Mäher auf jeden Fall bis zum ersten Punkt nach der Station nur mit ODO fahren, wenn "DOCK_IGNORE_GPS = true" ist.
Was hälst Du davon?

stephwe, 2021-09-26:
ja, das hört sich gut an und beide Funktionen arbeiten dann schön zusammen - falls ausgewählt. 
Heute noch rund 10 Mal die GPS Reset Funktion getestet, hat jedes mal einwandfrei gearbeitet, egal welches Signal (FIX, FLOAT, INVALID) der Mover vor dem reset hatte.
Eine kurze Wartezeit von vielleicht 10sek. nachdem der Mover FIX bekommen hat wäre noch gut, ist aber nicht zwingend erforderlich. Bei mir hat alles 100% geklappt.

Svol0, 2021-10-02:
Für den GPS-Reset beim undocking habe ich folgenden aktualisierten Vorschlag (alle Zeilennummernangabe beziehen sich auf den release Stand 1.0.219):
- Erweiterung der config.h mit folgenden Parametern:
```
#define DOCK_SLOW_ONLY_LAST_POINTS  2 // dockingpoint number (counted from last dockingpoint) where slow speed will be used to reach the dockingstation (0 = all points will be reached with slow speed)
#define DOCK_POINT_GPS_REBOOT       2 // dockingpoint number (counted from last dockingpoint) where the gps will be rebooted and waited for gps-fix by undocking. 0 = no gps reboot by undocking
```

- Erweiterung der robot.h bei Zeile 147:
```
extern int dockGpsRebootState;        // state for gps-reboot at specified docking point by undocking action
```

- Erweiterung der robot.cpp bei Zeile 1332 (vor `  if (mow)  {  // wait until mowing motor is running`):
```
  // reboot gps by undocking at a specified docking point (please see "DOCK_POINT_GPS_REBOOT" in config.h) //SOew
  if (dockGpsRebootState > 0){   // status dockGpsReboot: 0= off, 1= reset gps, 2= wait for gps-fix, 3= check for stable gps-fix
    switch (dockGpsRebootState){
      case 1:
        // reboot gps to get new GPS fix
        gps.reboot();   // reboot gps to get new GPS fix
        dockGpsRebootTime = millis() + 5000; // load check timer for gps-fix with 5sec
        dockGpsRebootFixCounter = 0;
        dockGpsRebootState = 2;
        CONSOLE.println("robot.cpp  dockGpsRebootState - start gps-reboot");
        break;

      case 2:
        // wait for gps-fix solution
        if (dockGpsRebootTime <= millis()){
          if (gps.solution == SOL_FIXED){
            dockGpsRebootState = 3;
            dockGpsRebootFeedbackTimer  = millis();
            dockGpsRebootTime = millis(); // load check timer for stable gps-fix
            dockGpsRebootDistGpsTrg = false; // reset trigger
            CONSOLE.print("robot.cpp  dockGpsRebootState - got gps-fix after ");
            CONSOLE.print(dockGpsRebootFixCounter);
            CONSOLE.println(" sec");     
          }
          else {
            dockGpsRebootTime += 5000; // load check timer for gps-fix with 5sec
            dockGpsRebootFixCounter += 5;  // add 5 seconds
            if (!buzzer.isPlaying()) buzzer.sound(SND_TILT, true);
            CONSOLE.print("robot.cpp  dockGpsRebootState - still no gps-fix after ");
            CONSOLE.print(dockGpsRebootFixCounter);
            CONSOLE.println(" sec");     
          }
        }
        break;

      case 3:
        // wait if gps-fix position stays stable for at least 20sec
        if ((gps.solution == SOL_FIXED) && (millis() - dockGpsRebootTime > 20000)){
          dockGpsRebootState = 0; // finished  
          CONSOLE.println("robot.cpp  dockGpsRebootState - gps-pos is stable; continue undocking;");
        }
        if (gps.solution != SOL_FIXED) dockGpsRebootState = 2; // wait for gps-fix again
        if (dockGpsRebootDistGpsTrg == true){ // gps position is changing to much
          dockGpsRebootDistGpsTrg = false; // reset trigger
          dockGpsRebootTime = millis();
          CONSOLE.print("robot.cpp  dockGpsRebootState - gps-pos is moving; timereset after");
          CONSOLE.print((millis() - dockGpsRebootTime));
          CONSOLE.println("msec");
          if (!buzzer.isPlaying()) buzzer.sound(SND_ERROR, true);               
        }
        if (dockGpsRebootFeedbackTimer <= millis()){
          dockGpsRebootFeedbackTimer = millis() + 5000;
          if (!buzzer.isPlaying()) buzzer.sound(SND_READY, true);
        }
        break;
        
[... 102 lines cut ...]
  }
  return false;
}
```

Funktionsbeschreibung:
- Bei Fahrt zur Docking-Station wird über den Parameter Wert von "DOCK_SLOW_ONLY_LAST_POINTS" die Position des Dockingpunktes angegeben (betrachtet aus Richtung Dockingstation), ab welchem mit langsamer Geschwindigkeit (linear = 0,1) weiter gefahren wird. Alle Dockingpunkte vorher werden mit der normalen (setspeed) Geschwindigkeit angefahren. Ein Wert von "Null" bewirkt, dass alle Punkte mit langsamer Geschwindigkeit angefahren werden.
- Bei Fahrt aus der Docking-Station wird über den Parameter Wert von "DOCK_POINT_GPS_REBOOT" die Position des Dockingpunktes angegeben (betrachtet aus Richtung Dockingstation), ab welchem ein GPS-Reboot durchgeführt werden soll. Der Mäher wartet dann, bis ein GPS-Fix vorhanden ist, und setzt das undocking fort. Da der Mäher dann eine korrekte Position hat, wird der Rest der Dockingstrecke vorwärts gerichtet, mit normaler Geschwindigkeit und GPS-Unterstützung fortgesetzt. Bei aktiviertem "DOCK_IGNORE_GPS" wird nur bis zum GPS-Reset Punkt ohne GPS-Unterstützung gefahren. Ein Wert von "Null" bewirkt keinen GPS-Reboot beim undocking.
Der Dockingpunkt für den GPS-Reboot sollte so gewählt werden, dass dieser sich an einer Stelle befindet, wo der Mäher einfach einen guten GPS-FIX bekommen kann.

Detailierter Ablauf bei Verwendung von "DOCK_POINT_GPS_REBOOT" in Kombination mit "DOCK_IGNORE_GPS" = true:
 - Mäher fährt rückwärts mit langsamer Geschwindigkeit und nur mit IMU/ODO bis zu dem bei "DOCK_POINT_GPS_REBOOT" eingestelltem Dockingpoint.
 - Dort angekommen wird ein GPS-Reset durchgeführt und der Mäher stoppt.
 - Während auf ein GPS-FIX gewartet wird, ertönt alle 5 Sek. ein kurzer Doppelton durch den Buzzer als akustisches Feedback.
[... cut ...]
Wer kein akustisches Feedback möchte, kann alle Zeilen die mit "if (!buzzer.isPlaying())" beginnen einfach auskommentieren.
Zur Zeit sind auch noch Consolen-Ausgaben vorhanden, um eine bessere Kontrolle der Funktion beim testen zu haben.

HINWEIS:
Dieses wurde bereits mehrfach erfolgreich getestet, trotzdem erfolgt die Verwendung auf eigene Gefahr. 
Die fertig geänderten robot.h, robot.cpp und map.cpp habe ich als ZIP angehängt. Die Dateien entsprachen vor der Anpassung dem Softwarereleasestand von 1.0.219.

Gruß Sven

stephwe, 2021-10-03:
Heute mehrfach getestet. Ausfahrt mit ODO bis Dockingpunkt 2 - Reset GPS - Warten auf FIX + 20sek stabiler FIX - Start Mähvorgang. Hat bei allen Versuchen ohne Fehler geklappt.
Einfahrt zur Dockingstation ebenfalls ab Dockingpunkt 2 langsam und auch zu 100% geklappt.

Akustisches Feedback konnte ich leider nicht testen da mein Buzzer defekt ist.

stephwe, 2021-10-03:
Nachtrag:
[GPS reboot.txt](https://github.com/Ardumower/Sunray/files/7273409/GPS.reboot.txt)

nach mehrmaligen Testen hat dann doch der Undockvorgang nicht mehr geklappt. Laut Monitor ist scheint das warten von 20 sek. auf ein stabiles Signal zu sensibel eingestellt.
Siehe angehängte TXT.
Der Mover hatte FIX aber anscheinend wurden minimalste Veränderungen des GPS Signals registriert und dementsprechend gab es max. 1sek lang ein stabiles GPS Signal.

Svol0, 2021-10-03:
Hi @stephwe,
Ok. Dann ändere bitte mal den Wert in der Robot.cpp  bei `if (distGPS > 0.02) dockGpsRebootDistGpsTrg = true;  // reset timer for solid gps-fix check (please see "dockGpsRebootState") //SOew`
von 0.02 auf 0.2.
0.02 ist wahrscheinlich wirklich zu sensibel. Als ich es getestet habe, hab ich mir den Wert von distGPS mit rausschreiben lassen und dieser lag meistens bei 0.01.
Danke fürs Testen.
Gruß Sven

stephwe, 2021-10-03:
Hallo Sven,

ja, jetzt klappt es. 
[GPS reboot.txt](https://github.com/Ardumower/Sunray/files/7274234/GPS.reboot.txt)
distGPS > habe ich jetzt auf 0.05 gesetzt und es funktioniert. Mehrmaligen Undock-Vorgang mit Reset des Movers versucht - jedes Mal geklappt.
Auch wirklich genial das man die Empfindlichkeit von distGPS einstellen kann. Wird ja bestimmt bei einigen anders sein wie bei mir oder dir.
Super Sache. 
Diese Woche komme ich leider zu nichts mehr aber am Sonntag werde ich weiter testen - sieht auf alle Fälle super aus.

stephwe, 2021-10-08:
Heute mehrere Test mit verschiedenen Dockingpoints gemacht. distGPS habe ich auf 0,4 gesetzt.
Hat alles 100% geklappt. 
Wartezeit nach dem Undocking war unterschiedlich aber nie eine falschen FIX bekommen.
Super ! Alle Daumen hoch.