Ardumower/Sunray #55: add maximum drive speed limitation in config.h

Svol0, 2021-10-09:
Since it has been possible to increase the maximum driving speed from 0.39m/sec to 0.59m/sec since a change in the Sunray app, I would like to extend the config.h with the following parameter:
`#define MOTOR_MAX_SPEED 0.30          // limitation for setSpeed value from Sunray-App (0,01 to 0,59m/sec are possible) to avoid too high speed setting by mistake `

Change the line `if (floatValue >= 0) setSpeed = floatValue;` in comm.cpp (line 79 release V1.0.219) into
```
          if (floatValue >= 0) {
            setSpeed = floatValue;
            if (setSpeed > MOTOR_MAX_SPEED) setSpeed = MOTOR_MAX_SPEED;   // limitation for setSpeed (please see "MOTOR_MAX_SPEED" in config.h) //SOew
          }
```

It has happened to me a few times out of carelessness that I accidentally set the slider on too high speed value. The drives could not reach this speed, which meant that the mower could no longer work correctly.
The maximum drive speed for the mower can be limited to a value that matches the system.

stephwe, 2021-10-10:
getestet mit Sunray 1.0.119 und 1.0.120 - funktioniert tadellos.
Danke.

MrTreeBark, 2022-05-30:
Mine is reaching a speed of 0.59 without problems.
The reason it could not work this way is:
The fixed stanley parameters. Use the code of Interpolation of parameters like i commited.

Anyway, this speed is to fast for mowing.. but perfect for a runaway becouse of heavy rain.