Ardumower/Sunray #59: Support arbitrary mower GPS antenna position

greymfm, 2021-12-19:
Currently, the map heading (alpha in degree) is calculated as follows. We should support an arbitrary GPS antenne position by:

1. Back-rotating the current mower rotation to the reference mower position (using IMU)
2. Computing the map heading (alpha in degree) as usual

[image]

ShadedSelf, 2023-08-10:
Ive been working on something similar, my idea was to use it to compensate for tilt, so the mower position would be evaluated at the bottom of the wheels, but it can be used to select any arbitrary virtual antenna position.

Its a bit messy, it would be better to use the quaternion we get dirrectly from the imu but I havent gotten around to do that, usgin a 6dof imu complicates things too.
Here is how it looks like:

 ```
//update heading first
 if ((imuDriver.imuFound) && (maps.useIMU)) {
    // IMU available and should be used by planner
    stateDelta = scalePI(stateDelta + stateDeltaIMU );          
  } else {
    // odometry
    stateDelta = scalePI(stateDelta + deltaOdometry);  
  }

  quat_t x; x.setRotation({1,0,0}, imuDriver.pitch, false);  
  quat_t y; y.setRotation({0,1,0}, imuDriver.roll, false);  
  quat_t z; z.setRotation({0,0,1}, stateDelta, false);
  quat_t rot = (x*y*z).norm();

  vec3_t forward = rot.rotate({1,0,0}, GLOBAL_FRAME).norm() ^ vec3_t(1,1,-1);
  vec3_t right = rot.rotate({0,1,0}, GLOBAL_FRAME).norm() ^ vec3_t(-1,-1,-1);
  vec3_t up = rot.rotate({0,0,1}, GLOBAL_FRAME).norm() ^ vec3_t(-1,-1,1);

  float posN = 0;
  float posE = 0;
  if (absolutePosSource){
    relativeLL(absolutePosSourceLat, absolutePosSourceLon, gps.lat, gps.lon, posN, posE);    
  } else {
    posN = gps.relPosN;  
    posE = gps.relPosE;     
  }   

  vec3_t tiltOffset = up * (-24.0 / 100.0);
  posN += tiltOffset.y;
  posE += tiltOffset.x;
```

Im using [this library](https://github.com/RCmags/vector_datatype) for vectors and  quaternions.

greymfm, 2023-08-10:
Nice. I have added quaternions to the IMU software driver (https://github.com/Ardumower/Sunray/blob/f058a6ad426ca10aaafe206af83b0b5f878f86c5/sunray/src/driver/RobotDriver.h#L101), so you can now access IMU quaternion like this:
imuDriver.quatW, imuDriver.quatX, imuDriver.quatY, imuDriver.quatZ