Ardumower/Sunray #61: Problem compiling

jonaslindemann, 2022-02-14:
Hi!

I get this when compiling version 1.0.230:

Sunray-1.0.230\sunray\src\mpu\inv_mpu.c:53:1: error: no return statement in function returning non-void [-Werror=return-type]

Seems to be related to an empty

static inline int reg_int_cb(struct int_param_s *int_param)
{
	
}

Thanks in advance.

greymfm, 2022-02-15:
Hello, can you send me you config.h file? Looks like some definition from config_example.h might be missing...

jonaslindemann, 2022-02-15:
Here it is:

https://www.dropbox.com/s/46jtpn81niaq9ue/config.h?dl=0

We have the standard Ardumower kit. Board 1.4

greymfm, 2022-02-15:
Thanks, no issues (using your config.h) with compilation here. I'm on Arduino IDE 1.8.13. Which one did you try?

jonaslindemann, 2022-02-15:
I downloaded the latest version, 1.8.19, perhaps I should have downloaded 1.8.13.

jonaslindemann, 2022-03-05:
Problem solved. Used the wrong version of a library.

closed 2024-03-13 by greymfm