Ardumower/Sunray #62: ESP32_BLE and Sunray

jonaslindemann, 2022-03-05:
Hi!

I am not sure support for the ESP32 is working the latest firmware .230. It seems it references code for the HM-10 module. Serial output from the main board says it can't find Bluetooth.

Jonas

jonaslindemann, 2022-03-11:
Anyone have an idea what can be the problem?

redprobe, 2022-04-17:
I think I've had the same problem. I also added a thread on the forum, got no reply yet.
For me, when I boot the mower, the Arduino get's stuck saying it can't find the ESP8266.
After that the Arduino reboots and nothing happens.

I've downgraded the mower to an earlier version, which made it work again.

stilsch, 2022-04-26:
mine (PCB 1.3) is working with ESP32 and .230 (but is also arguing about missing ESP8266 during startup)

cdego, 2022-04-28:
i have the same problem. One time, it was ok, but just one time.
i will try to downgrade.

jonaslindemann, 2022-05-19:
I still have the problem with the ESP32. Can't connect to Bluetooth from the main board. I am not sure what @cdego means with http/https is that in the config.h file?

cdego, 2022-05-19:
For wifi, we have to write http, without s in the adress bar. Nothing to do in config.h

jonaslindemann, 2022-05-19:
Ok, so my problem still persists. I can connect to Bluetooth, but no information is transferred. The main board does not find the Bluetooth module. I will debug more tonight.

et05hk6, 2022-08-23:
This was a bit old, but if you haven't solved it... The ESP32 is alive and connects to wifi/bluetooth, but it seems it can't talk to the Arduino-board! I would say double and triple-check the serial connections and Sunray configurations! If you have a logic analyzer available then it would be time to bring that up (and if not, then go buy one! I use a Saleae pirate-copy from ebay for like $5 together with the software from Saleae. Worth every penny in cases like this!)