Ardumower/Sunray #66: Fehlerhafte Routenplanung beim skippen

Svol0, 2022-04-29:
Hallo @greymfm,
bei der Berechnung des nächsten Wegpunktes scheint es einen Fehler zu geben, wenn man den Schieberegler "skip to progress(%)" im laufenden Mähbetrieb betätigt. Nach Betätigen des Schiebereglers wird der nächste Wegpunkt in direkter Line ohne Berücksichtigung von Exklusionen und Perimeter angefahren. Ich habe hier ein Beispiel, wie er versucht auf direktem Weg den Punkt anzufahren.
[image]

Getestet habe ich dieses Verhalten allerdings mit der Master 239. Ob es in der 230 Release ebenfalls auftritt habe ich nicht getestet.
Wenn Du zusätzliche Informationen benötigst, lass es mich bitte wissen.

Gruß Sven

greymfm, 2022-05-06:
Hallo Sven,
danke Dir vielmals für das Feedback! Der Bug sollte ab v1.0.247 nun behoben sein ( https://github.com/Ardumower/Sunray/commit/a7ad03da1397f1cfa32a69e7afbbfbf8641b27fc )
Gruss,
Alexander

Svol0, 2022-05-06:
Hallo Alexander,
danke für die schnelle Behebung.
Gruß Sven

Svol0, 2022-05-15:
Hallo Alexander,
ich konnte es leider erst heute in der 1.0.249 testen.
Die Berechnung des nächsten Wegpunktes beim skippen funktioniert jetzt ordnungsgemäß.
Vielen Dank.
Gruss,
Sven

closed 2022-05-15 by Svol0