Ardumower/Sunray #68: Dynamic Mowerspeed and Dynamic Mowmotor RPM

MrTreeBark, 2022-05-13:
Hello.

I tried the Dynamic Mowing Speed and RPM function in .230.

Unfortunately, i think it is the wrong procedure/method, It doesnt work well, and has Bugs.
Its of no use for me.

The Mowermotor should have a PID Controller just like the Drivemotors.
The control loop measurement variable should be the actual RPM over ticks per revolution like the drive motors.
This way you have no dead time in Control Loop, and control accordingly to Setpoint and Actual value,,, And let the Driver do its thing,
The Speed of the mower should be linked to weather the controller is able to regulate the RPM of the Mowermotor 
to the Setpoint.  The RPM at the Blade is directly connected to the Speed, as soon the Blade slows down, the Mower will too and eventually stop and back up for 5cm. That is the only smart way to do it.
Do not manipulate PWM after the functions calculated them, that disturbs already fine working functions in your code.
The controller should affect the linear values of the different moving operation, thats all. No PWM manipulation.
Please remove this feature, it doesnt work well and it´s messing up the code.

greymfm, 2022-05-22:
Hey, Yes, this code should be removed (actually, it was someone-else pull-request that I did not fully understand...)
Also, I'm currently heavily refactoring the robot.cpp code (so that things get more sorted...)

MrTreeBark, 2022-05-28:
Great, thanks for your work.

fracii, 2023-06-11:
I'm have tried latest version and I think the old pull request for Dynamic mow motor was working much better than actual latest release. Now the mow motor stops to easy and robot does not slow down enough.Any quick fix to give mow motor more power so it does not directly stop when little higher grass?