Ardumower/Sunray #76: Mower undocks automatically at full charge after manual docking

Svol0, 2022-06-12:
If the setting DOCK_AUTO_START is activated and the mower is send during mowing action to the docking station manually (Dock button), the mower automatically starts mowing again if the battery is fully charged.
Normally the mower should only go on with mowing, if he was send automatically to the docking station in case of empty battery. If the mower is send manually or by rain sensor to the docking station, the mower should stay there until he gets the external start command.

I have recognised the issue in master V1.0.274.

regards,
Sven

stephwe, 2022-07-10:
Problem tritt bei mir auch auf. Beim manuellen Docking und nicht fertig gemähtem Mähplan fährt der Mover nach Beendigung des Laden selbstständig wieder raus.
Dock durch Regensensor funktioniert und Mover bleibt bis zum manuellen Start in der Ladestation, auch bei Beendigung der Ladung.

greymfm, 2022-07-15:
Sollte behoben sein ab v1.0.284  :-)

https://github.com/Ardumower/Sunray/commit/e68daddb4c201a90511387a4b535656d4e78a855

Svol0, 2022-07-25:
Hallo @greymfm ,
ich konnte es heute testen und es funktioniert.
Vielen Dank.
Gruß Sven

closed 2022-07-25 by Svol0