Ardumower/Sunray #135: Compile Error

MrTreeBark, 2023-07-12:
src\ublox\ublox.cpp: In member function 'void UBLOX::dispatchMessage()':
\src\ublox\ublox.cpp:478:51: error: expression cannot be used as a function
  478 |                       rmin = min(rmin, fabs(prRes));
      |                                                   ^
Mehrere Bibliotheken wurden für "Adafruit_ZeroDMA.h" gefunden
 Benutzt: C:\Users\seppb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.3\libraries\Adafruit_ZeroDMA
 Nicht benutzt: C:\Users\seppb\Documents\Arduino\libraries\Adafruit_Zero_DMA_Library
exit status 1
Fehler beim Kompilieren für das Board Adafruit Grand Central M4 (SAMD51).

There is this compilation error i get. If i remove the timevariables in Robotdriver.h for the gpsdriver class the issue is gone.
Is that my compilerversion? Cant compile any versions of sunray after GPS time variables had been added.

greymfm, 2023-07-12:
Try this :-) (commited as compilation fix today)

                      rmax = ::max(rmax, fabs(prRes));
                      rmin = ::min(rmin, fabs(prRes));

To access any functions defined in the global scope, when there is a function defined with the same signature...

MrTreeBark, 2023-07-12:
Worxs!
Solved.

closed 2023-07-12 by MrTreeBark