Ardumower/Sunray #153: compilation error: undeclared 'SIOCGSTAMP_OLD'

kraiz, 2024-03-28:
I get a compilation error that seems to relate to a change made 2 days ago:

```
[ 39%] Building CXX object CMakeFiles/sunray.dir/home/pi/Sunray/sunray/i2c.cpp.o
[ 40%] Building CXX object CMakeFiles/sunray.dir/home/pi/Sunray/sunray/ioboard.cpp.o
[ 41%] Building CXX object CMakeFiles/sunray.dir/home/pi/Sunray/sunray/linuxcan.cpp.o
/home/pi/Sunray/sunray/linuxcan.cpp: In member function 'virtual bool LinuxCAN::run()':
/home/pi/Sunray/sunray/linuxcan.cpp:110:14: error: 'SIOCGSTAMP_OLD' was not declared in this scope
  ioctl(sock, SIOCGSTAMP_OLD, &tv);
              ^~~~~~~~~~~~~~
CMakeFiles/sunray.dir/build.make:1526: recipe for target 'CMakeFiles/sunray.dir/home/pi/Sunray/sunray/linuxcan.cpp.o' failed
make[2]: *** [CMakeFiles/sunray.dir/home/pi/Sunray/sunray/linuxcan.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/sunray.dir/all' failed
make[1]: *** [CMakeFiles/sunray.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

https://github.com/Ardumower/Sunray/blame/68ea97119b7f867dfc641d85c0ae730c906b8071/sunray/linuxcan.cpp#L110

I have not much experience in C/Make/... - I'm just compiling to get the MQTT credentials included - but just want to report it.

greymfm, 2024-03-28:
Thanks! Should be fixed in this commit now :-)

https://github.com/Ardumower/Sunray/commit/ebe869683ead6222decb0f0d5e76760aaf9cb02a

kraiz, 2024-03-29:
It is, thx.

closed 2024-03-29 by kraiz