Ardumower/Sunray #160: Obstacle -> back to route, violates the perimeter

themanfrommoon, 2024-06-09:
[image]
Ich machs mal kurz: Wann kommt's?

Beste Grüße,
Chris

greymfm, 2024-06-16:
Actually, this is a good enhancement and we should give priority to it, so it will be implemented as soon as possible. Meanwhile I have written the requirements in English language, so that a human (or AI) can understand what to do for the implementation:

Context:
A differential-drive robot mower uses RTK GPS to mow the lawn by a given list of waypoints (coverage planning). In addition to the waypoint list, it has the following data: a perimeter (a list of points that describes the waypoints that the robot center point can follow around the working area), a list of exclusions (each exclusion describes the waypoints that the robot center point can follow around no-go areas) and a list of dynamic obstacles (each obstacle describes the waypoints that the robot center point can follow around the obstacle).

Requirement:
Add code so that half of the robot width, given by the robot footprint (4 points that the describe the boundary of the robot and a center point where the robot differential wheels turn around) never crosses any perimeter line, any obstacle line, any exclusion. Instead additional waypoints should be inserted for the next waypoint in the list.

greymfm, 2024-06-16:
Further thinking: this will also have an influence on the ordinary mowing process (U shape pattern mowing) as half of robot width also crosses the perimeter lines there... actually this idea requires more thinking