Ardumower/Sunray #161: New "LPF" filter for PID Conrolers in motor.cpp producing "NaN´s" after some (long) time.....

MrTreeBark, 2024-07-21:
What the description said, possibly due to an overflow. Observed a long time and validated with console, mower conrol loop runs with 50Hz. Mower will stop after ~ 1 - 2  hours and drives are dead, because pid is outputting 0, beside rpm set is given.

MrTreeBark, 2024-07-22:
Update: it may give Nan´s after stopImmidialetly, i don´t know why...

ShadedSelf, 2024-07-22:
I think its because stopImmediately resets control loop time inside pid, "Ta" then becomes 0 and you get division by zero.

This is a quick fix I did on PID.cpp:

Ta = max(Ta, 0.000001);

MrTreeBark, 2024-07-24:
This is a good hint, but why? Ta is computeted inside PID, its just the delta time, why should it stop working just like that and not recover, because cyletime would be maybe 0 for the pid on a reset, but just for one iteration? somehoe it dies and keeps to be dead, even when mower is rebootet.

MrTreeBark, 2024-07-25:
And something else, i read about those microcontrollers... this extensively using of floats for absolutely no reason but reading for human mind is not a good thing to do. cm, meters etc.. times ... everything should be integers and translated only in console for readability. all this floats are just eating away on computeprecision and hardware recources, nothing else.

ShadedSelf, 2024-07-25:
Actually, i think this its just the LPF, not the PID, the PID controller might actually recover because it gets casted to int (?).

Might actually come from here:
https://github.com/Ardumower/Sunray/blob/637bcb6b97d278cd2d3652bcaeb4f8eaace84ba7/sunray/lowpass_filter.cpp#L23
Then y and y_prev become NaN.

I dont have the LPF on my version so I cant say for sure.