Ardumower/Sunray #163: Provide DEB Package and APT Repo

schrieveslaach, 2024-08-24:
Currently, I'm trying to get my Alfred up and while doing so I'm running into issues (and still no success mowing the lawn). One thing that bogs me is the software handling process. For example, images are provided via HTTP (not HTTPS) or the update process via sunray doesn't work (get some cURL error). Also, the image provided on the preconfigured SD card was running a Linux 4.x kernel.

These things worry me from my software development background (software integrity, usability, etc.) and I want to help to make that better. 

Having an [APT repository a.k.a. Debian Repository](https://wiki.debian.org/DebianRepository) for the Sunray software would improve these aspects. For example, there are ready made integrity checks and also improves the installation process.

I imaging that one should be able to take a stock RaspberryPI OS and adds an entry to the [APT sources](https://wiki.debian.org/SourcesList) and the Sunray application can be installed and enabled with following commands:

```bash
# add the entry to APT sources
apt update
apt install sunray-alfred
systemctl enable sunray
systemctl start sunray
```

To save one from maintaining the APT repo and hosting, one could use some GitHub actions and hosts it on GitHub pages (like shown [here](https://github.com/marketplace/actions/create-apt-repository)).

I'm willing to help in this process. Please, let me know what you think.

greymfm, 2024-08-28:
I like the idea... but what about custom robot builds - 99% of my robots (Alfred, modified Alfred with LiDAR, owlMower XL, ...)  use custom sunray builds (with custom config files, so custom binaries) - how does that fit here...

schrieveslaach, 2024-08-28:
When I compare some of the config files in the [alfred directory](https://github.com/Ardumower/Sunray/tree/e24b238d60b68b8ea6846a9a1c1f743a5c789346/alfred) it appears to me that these C header files could turned into actual config files that could live in the `/etc/sunray` directory and be more compliant with the [Linux Filesystem Hierarchy Standard](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard). I could imaging that there is an initial setup Guide that let's you configure this ().

Meanwhile I experimented with GitHub actions in a [personal fork](https://github.com/schrieveslaach/Sunray) and I was able to build a DEB package with CI (see [here](https://github.com/schrieveslaach/Sunray/actions/runs/10586293651)). 

Unfortunately, I discovered that the Banana PI M4 seems to be incompatible with Debian based distros (no Raspberry PI OS, armbian or dietpi compatible) and it appears to me that this board has no good Linux support. Can you explain why the Banana PI M4 is the default choice for Alfred RKT?

Are you open to have a video chat about the topic? There are also security issue I see in the [Sunray webapp](http://grauonline.de/alexwww/ardumower/ardumower-sunray-app/index.html)  that worry me too.