Ardumower/Sunray #173: ublox parsing issue, memory leak/Infinite consumption

MrTreeBark, 2025-06-08:
The unparsed Message in function parse in ublox.cpp leads to infite growing String unparsedMessage If ublox picks something Up which is Not starting with expected chars:

`void UBLOX::parse(int b)
{
  if (debug) CONSOLE.print(b, HEX);
  if (debug) CONSOLE.print(",");
  
  if ( (this->state == GOT_NONE) || (this->state == GOT_SYNC1) ) {
    char ch = char(b);
    if (ch == '$') unparsedMessage = "";    
    **unparsedMessage += ch;**
    **if ((ch == '\r') || (ch == '\n'))** {
      //CONSOLE.println(unparsedMessage);
      if (unparsedMessage.startsWith("$GNGGA")) {
        nmeaGGAMessage = unparsedMessage;
        nmeaGGAMessage.trim();
      }
      unparsedMessage = "";
    }
  }`

Solution:
Just clear the Message that is Not parsed:

`void UBLOX::parse(int b)
{
  if (debug) CONSOLE.print(b, HEX);
  if (debug) CONSOLE.print(",");
  
  if ( (this->state == GOT_NONE) || (this->state == GOT_SYNC1) ) {
    char ch = char(b);
    if (ch == '$') unparsedMessage = "";    
    unparsedMessage += ch;
    if ((ch == '\r') || (ch == '\n')) {
      //CONSOLE.println(unparsedMessage);
      if (unparsedMessage.startsWith("$GNGGA")) {
        nmeaGGAMessage = unparsedMessage;
        nmeaGGAMessage.trim();
      }
      unparsedMessage = "";
    } **else {
      unparsedMessage = "";
    }**
  }`

greymfm, 2025-06-08:
Suggested solution would never parse any NMEA message as it always resets it if the ending character is missing?

[UBX message][UBX message][NMEA message][UBX message][UBX message][NMEA message]...

greymfm, 2025-07-19:
Thanks . Solved by this I assume:

[image]